DEWETRON/OXYGEN-SDK #3: How to Debug a Plugin

HpLightcorner, 2020-07-03:
Hey Dewetron Team,
I am developing a custom plugin and so far everything works perfectly (also due to your well down wiki).

Without working through your plugin-loading way, is there a simple and efficient way to debug a plugin? (and I mean true debugging inside Visual Studio - of course I am already using log files)...

If so, It would be great to add some documentation.
Kind regards

glaure, 2020-07-06:
One way is to start Oxygen using Visual Studio.

[image]

Or you can attach to a running Oxygen process:

[image]

Then you can set breakpoints in your plugins source code.

Best regards
Gunt

HpLightcorner, 2020-07-06:
Hey,
I will immediately give it a try - and let you know if everything works as expected.

HpLightcorner, 2020-07-06:
Works perfectly -thanks in advance!

closed 2020-07-06 by HpLightcorner