DEWETRON/OXYGEN-SDK #4: How to add a "Plugin-Level" Config

HpLightcorner, 2020-07-03:
Dear Dewetron Team,

My Plugin needs a "Plugin-Level" Config Input (in my special case a simple File-Path), similar to the ethernet-receiver:
[image]

There are now several questions arising:
- How can I implement such a config-parameter on "Plugin-Level"?
- If adding multiple instances of this plugin, will the current configuration/path stay the same when I hit the plus button to add a new channel?

Use-Case:
My Plugin provides n-channels, which are polled from a custom data source. Therefore, the plugin starts to spin a single thread for all channels as soon as a the first channel is getting active. Each Channel-Instance uses the data from the single polling thread. However, the polling-thread itself has to be configured using an situation-specific YAML File - and of course I would like to use the same configuration for each instance of the plugin.

jpoellitsch, 2020-07-06:
SoftwareChannelInstances are rather meant to be self contained and independent from each other (one instance per independent data source).

However, its possible to provide multiple output channels out of one SoftwareChannelInstance and all of them can have their own configuration parameters too.

The resulting workflow would be: 

- on SoftwareChannelInstance::create() : add the plugin configuration properties you need (path to the yaml file) to the root channel

Therefore, on creation, only an "empty" instance group channel gets created in oxygen.
But when path of yaml file is entered (or changed afterwards) in the detail setup of this channel in oxygen, SoftwareChannelInstance::update() is called:

- on SoftwareChannelInstance::update() you can read out the value of the configuration file, parse it and add as many output channels as needed via addOuputChannel()
(they will automatically get displayed in oxygen as sub-channels of the plugin instance root channel)

Would that be a feasible solution for your problem?

HpLightcorner, 2020-07-06:
Hey,
That sounds feasible. I will develop a prototype and come back if the solution does work for me.
Shall I leave the issue open until I can give you some feedback on the approach?

HpLightcorner, 2020-07-07:
Hey,
The first prototypes are promising. I have some further questions regarding the approach:

From my understanding, channel 0 - the root channel, is never going to sample anything, hence I should set Sample-Format to `odk::ChannelDataformat::SampleOccurrence::NEVER`. When I do so - my plugin does not show up:

```
    void create(odk::IfHost* host) override
    {
        getRootChannel()->setDefaultName("Test-Channel-Root")
            .setSampleFormat(
                odk::ChannelDataformat::SampleOccurrence::NEVER,
                odk::ChannelDataformat::SampleFormat::DOUBLE,
                1)
            .setDeletable(false)
            .addProperty(KEY_INPUT_FILE, _input_file);
    }
```

Do I miss here something? The Root-Channel simply groups the other channels together, but will never process data, so sample format would then be correct?

And then another advice would be kind: I do sampling for all Output-Channels in a second thread (as you suggested, adding them using `addOuputChannel()` and update them in the process-callback. I set them to `odk::ChannelDataformat::SampleOccurrence::SYNC`, and when process is called I add as many samples as I have since the last call of process. But what shall I do if I my acquisitor-thread is too slow and I cannot sample fast enough - or in other words, if I am out of sync? Shall I fill in dummy-values until I am "synced" again? Situation where this can happen: The sampling source gets disconnected and connected again (hot-plug).

HpLightcorner, 2020-07-07:
What actually works as expected:
```
    void create(odk::IfHost* host) override
    {
        getRootChannel()->setDefaultName("Test-Channel-Root")
            .setDeletable(false)
            .addProperty(KEY_INPUT_FILE, _input_file);
    }
```

Do not call `setSampleFormat` at all for the root channel. But this somehow feels not "correct" if there is an explicit "never" condition.

jpoellitsch, 2020-07-08:
> Hey,
> The first prototypes are promising. I have some further questions regarding the approach:
>
> From my understanding, channel 0 - the root channel, is never going to sample anything, hence I should [...]

The root channel can be treated like any other output channel either providing channel data or not (In addition to the grouping functionality).

But you discovered a little glitch in the framework, handling the illegal combination of SampleFormat::DOUBLE and SampleOccurrence::NEVER not very graceful. We will try to make that more clear in the future.

If you dont want the root channel providing samples, SampleFormat::None should be used in combination with SampleOccurrence::NEVER:

root_channel->setSampleFormat(
            odk::ChannelDataformat::SampleOccurrence::NEVER,
            odk::ChannelDataformat::SampleFormat::NONE,
            0);

But you dont have to, because that is the default anyway.

> And then another advice would be kind: I do sampling for all Output-Channels in a second thread (as you suggested, adding them using `addOuputChannel()` and update them in the process-callback. I [...]

There is currently no support for "gaps" in synchronous channels, therefore you can only fill it with dummy values (NaN would be possible for floating point channels).
Only ASYNC channels can be freely distributed over time as (timestamp, value) pair, but that should not be used to store synchronous channels as it comes with an performance penalty.

HpLightcorner, 2020-07-09:
Hey,
Everything works perfectly now - i had to investigate once more on my sampling process. 
Also, the approach with the root channel works perfectly, hence I will close the issue.

Thanks for support!

closed 2020-07-09 by HpLightcorner