DEWETRON/OXYGEN-SDK #7: Initializing std::uint32_t with -1 might lead to compile-time errors

HpLightcorner, 2020-07-20:
**Issue:**
Depending on the compiler settings, initializing a varialbe with -1 might lead to compile-time errors (e.g. using MSVC2019 Toolchain: C2397). I found the issue in the following file:

at line 75: https://github.com/DEWETRON/OXYGEN-SDK/blob/c87c72d0bc3653ab55d76d1683730a42bd30bb78/odk/api/inc/odkapi_channel_dataformat_xml.h#L75

**Proposal**
Initializing the variable like this has the same effect but does not lead to compile-time errors:

`std::uint32_t m_sample_dimension = 0xFFFF FFFF;`

HpLightcorner, 2020-07-28:
I found that there is a valid-check in an example given: 
https://github.com/DEWETRON/OXYGEN-SDK/blob/c87c72d0bc3653ab55d76d1683730a42bd30bb78/examples/bin_detector/odkex_bin_detector.cpp#L177

Hence, I suggest to initialize with the following code:
`std::uint32_t m_sample_dimension = 0;`

as zero-dimension is invalid. I also forked the repository to implement and test it.

matthiasstraka, 2022-03-08:
Issue appears to be fixed

closed 2022-03-08 by matthiasstraka