DEWETRON/OXYGEN-SDK #9: Channels SDK recording

arnaudDewetronfrance, 2021-03-01:
I am currently working on a plugin for serial decoding. Channels are correctly displayed but not recording. 
Any idea what can lead this issue ? 
I can forward all necessary tools to simulate or reprouce it.

glaure, 2021-03-01:
Hi!

Is your plugin similar to this one?
[OXYGEN-SDK-SerialCsv](https://github.com/DEWETRON/OXYGEN-SDK-SerialCsv)

Can you please provide a screenshot of your ChannelList, so wie can have a look at the "Stored" property.

Best regards,
Gunther

arnaudDewetronfrance, 2021-03-01:
> Attention: This message comes from an external sender. Please be careful before opening links or attachments in this mail!

Hi Gunther ,

I hope that you are well.

No it is a totally different plugin developed from scratch.

Regards

glaure, 2021-03-01:
Arnauld, so you are seeing the plugins channels in the channel list? And the channels show values other than NAN?
But if you press record, the resulting file does not show the values.

Yes please provide source code to reproduce the issue.
And please attach your the plugin binary.

Regards,
Gunther

arnaudDewetronfrance, 2021-03-01:
Hi Gunther ,

Channel are seen but not recorded.

Yes but I need also to send you the simulator, I will prepar it.

Regards

arnaudDewetronfrance, 2021-03-01:
Hi Gunther ,

I have a question about the chrono icon that is in the right corner of the numeric instrument.

This icon is black , orange and then red. What is the meaning of this ? I don’t find infos on the internet page.

Regards

glaure, 2021-03-02:
Hi Arnaud,

this icon indicates how old the last shown value is. This indicates that the timestamping of your samples is not correct.

Do you use  getMasterTimestamp(host) ?

Regards
Gunther

arnaudDewetronfrance, 2021-03-02:
Hi Gunther ,

Here the source code and the simulator to generate serial frame.

getMasterTimestamp(host) is used to init the timestamp channel.

Regards

glaure, 2021-03-02:
Hi Arnaud,

github stripped the attachment. Please use my company email address directly.

Regards 
Gunther

glaure, 2021-03-02:
The function signature of addSample is:
inline void addSample(odk::IfHost* host, std::uint32_t local_channel_id, std::uint64_t timestamp, const void* data, size_t data_size);

std::uint64_t timestamp has to be a "tick". One way to set it is:
timestamp = getMasterTimestamp(host).m_ticks.

This has to be done for every call into process (in effect for every received sample).

Another issue is: .setSimpleTimebase(100000)
which means, that you would have to re-calculate each timestamp for this expected Timebase. I would remove this which means  using the resolution of the master timebase.

Btw:  Your name sample_index is wrong, because your channels are asynchronous.

arnaudDewetronfrance, 2021-03-02:
Hi Gunther ,

Thank for the quick feedback .

Regards