DEWETRON/OXYGEN-SDK #10: How to use “Log into Oxygen Logfile”？

anka120, 2022-01-11:
Since ‘Oxygen 6.0’ has been published.I wonder how to use the fuction of "Log into Oxygen Logfile"?Thanks.

matthiasstraka, 2022-01-11:
There is a special message for logging in the ODK. You need to send an `odk::host_msg::LOG_MESSAGE` message (either sync or async) with an `odk::IfStringValue` parameter to the host interface. The `key` parameter is a severity level with fatal = 1, error = 2, warning = 3, info = 4.

For example, when you have an `IfHost* host`, you can log an Info (4) text to it using the following code:
```cpp
auto log_msg = host->createValue<odk::IfStringValue>();
log_msg->set("Example log message async");
host->messageAsync(odk::host_msg_async::LOG_MESSAGE, 4, log_msg.get());
```
or
```cpp
auto log_msg = host->createValue<odk::IfStringValue>();
log_msg->set("Example log message sync");
host->messageSync(odk::host_msg::LOG_MESSAGE, 4, log_msg.get());
```
with a sync message call.

We will add this information to the documentation if it is not already there. Also, we will provided an enum/define for the severity level which appears to be missing in the GitHub repository.

matthiasstraka, 2022-01-11:
I added a new header `odk/api/inc/odkapi_logging.h` to the repository. It is not available in the Release 6.0 version but can be copied there manually if needed. In later releases, it will be included automatically.

This allows to write the severity level as an enum:
```
#include "odkapi_logging.h"
// ...
host->messageAsync(odk::host_msg_async::LOG_MESSAGE, odk::LOGLEVEL_INFO, log_msg.get());
```

Please let me know if your issue can be resolved now.

closed 2022-01-11 by matthiasstraka

anka120, 2022-01-11:
Yes,it works well.Thanks.
Another quesions:
1.Could I add notice like this,when customer edit the config in function of 'update()'?
[image]

matthiasstraka, 2022-01-11:
Unfortunately, this feature is currently not included in Oxygen 6.0. There has been preliminary work on status messages for a future Version of Oxygen, so you can expect it to work in Oxygen 6.1 or later. I will create a new GitHub Issue so we can track progress on it.

anka120, 2022-01-12:
Ok，thanks

commit 2f427c00 (2022-01-11): Add logging severity level enum (fixes #10)