DEWETRON/OXYGEN-SDK #25: FFT channels access in SDK

arnaudDewetronfrance, 2023-06-09:
We have a problem to access to the FFT channel ( PHA and PEAK )
The problem arises when we try to read them. 
We follow the next sequence:
- Get and Store the ids of the Base FFT channel (Complex), and the ids of the subchannels (AMP; PHI and Peak). So far, so good.
-  When we try to get the iterators for each channel using their IDs, we can only get a valid iterator for the base channel. 

The subchannel iterators for AMP, PHI and Peak that we get are invalid and when we use them to read the data, instead of getting a valid pointer to the channel data, we get a nullpointer.
In resume, we can get the subchannel channel ids, but we can not read them since we don't get a valid iterator or a valid proxy.

matthiasstraka, 2023-06-09:
We tried to replicate your findings using the [bin_detector](https://github.com/DEWETRON/OXYGEN-SDK/tree/master/examples/bin_detector) example plugin which can work with array channels such as the ones you described. 

This is our setup:
* Start Oxygen and create FFT calculation on any AI input channel
* Enable FFT output channels: Amplitude, Phase and Overall peak
* Select the `FFT_Amp` channel and create an `Example Plugin: Bin detector` calculation on the selected channel and verify that data is produced by the plugin without errors.

No matter which input channel we use (AMP, PHI, Peak), we always get a working calculation with valid input channel iterators. The following code gets the iterator (assuming input channels have been initialized correctly):

https://github.com/DEWETRON/OXYGEN-SDK/blob/1e3e856594f794415fa64287323d3ebf6f3ad061/examples/bin_detector/odkex_bin_detector.cpp#L257-L262

Can you please provide us a code snippet of how you retrieve and use these iterators in your code so we can replicate the problem and find a solution?

arnaudDewetronfrance, 2023-06-12:
> This email originated outside of DEWETRON. DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

Dear Matthias,

Thank for the answer.  It appear that when we tried to access to the different channels ( AMP and PHI ) , there is an issue but it is finally not a problem , by using 2 instance ,we have the expected behavior.

We have seen something strange in the FFT channels that affect the plugin on going that is related to the parameter as SR of the subchannel AMP and PHA.

I believe that the plugin that calculates the FFT has some issues as it doesn't update the sample rate/line resolution of its output channels until there is a change in its input signal (e.g., sample rate).
Our plugin reads the output channel parameters of the FFT to calculate the positions of the frequency components.
This can be observed when we change the "data size" parameter of the FFT. When we do that, the sample rate of FFT/Complex and the sample rate of the AVG plugin changes, but not the sample rate of Amplitude. However, if we then change the sample rate of the input signal and return it to the original value, we can see that the sample rate of FFT amplitude changes and becomes the same as the sample rate of FFT/Complex.

Let's see this with images

1. Start point. FFT data size = 1024. Everything it's ok
[image.png]
2. We change FFT Data Size to 2048. FFT/Complex Sample rate changes, AVG plugin sample rate changes, but FFT/Amp doesn't. Here we see the differences when we use the Spectrum analyzer between FFT and Average.
[image.png]

3. We change the sample rate of the INPUT from 100KHz to 1KHz and again to 100KHz and everything goes right again. Frequency components are shown in the correct position.
[image.png]

My conclusion is that the FFT/Amplitude channel has an issue updating it's parameters.

So the workaround is to change the SR of the channel and rechange it to the initial value to access correctly to the parameter value to make a right computation into the plugin.

Regards

https://github.com/DEWETRON/OXYGEN-SDK/blob/1e3e856594f794415fa64287323d3ebf6f3ad061/examples/bin_detector/odkex_bin_detector.cpp#L257-L262