DEWETRON/OXYGEN-SDK #37: Oxygen closing automatically after adding the new plugin

ravivelugula, 2024-10-14:
Hi

I have developed new torque plug and it is successfully added to oxygen software. How after adding input channel, the oxygen software is closing automatically by showing like below. 

Kindly provide me the possible solutions to this problem

Thanking you
[image]

glaure, 2024-10-14:
Hi,

Your plugin likely crashes the Oxygen process. 
You should have a look at the "Debugging and Logging" chapter in the SDK documentation:

https://dewetron.github.io/OXYGEN-SDK/20_DebuggingLogging/DebuggingLogging.html

Start Oxygen in the Visual Studio Debugger to have a look at the location where the crash occurs.

Bye Gunther

ravivelugula, 2024-10-21:
Hi 
Thank you prompt reply.

I am very new to this coding and debugging and I am unable to find the options in the debugging chapter (like unable to get property page option to "Change the Command from $(TargetPath) to the Oxygen.exe path" . Could you provide more step by step details? or a video ?  

Thanks and Regards

ravivelugula, 2024-10-21:
Hi 
Please see the below screenshot.
[image]

ravivelugula, 2024-10-21:
Hi 

English is fine for me. The problem is not getting options that are given in the debugging procedure. See the attached screenshot. There the property list is empty.

Thanks
[image]

Ravi

ravivelugula, 2024-11-13:
Hi,

Thank you for the previous solution. However, despite following the documented methods, I am still unable to debug the code successfully. I am unable to make further progress. If possible, I am open to sharing my plugin code with the GitHub team to help resolve the issue.

Thank you

ravivelugula, 2024-11-14:
Hi, issue is resolved. There problem in the plugin code itself.

closed 2024-11-15 by matthiasstraka