DEWETRON/OXYGEN-SDK #39: Sum of the three channels

ravivelugula, 2024-11-15:
Hi 

I have modified the code of sum of two channels for developing sum of three channels. The following the are modifications that I have made while developing the plugin:
1. m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN()})
2.   if (current_channel_ids.size() != 3) and at all the loops ( replced '2'  with '3').

However, the plugin file generated successfully , the oxygen is cashing and closing automatically once I select three input channels.

I would like know the solution for this issue.

Thanks
Ravi

glaure, 2024-11-15:
These are the necessary changes to sum three channels:

```
@@ -49,7 +49,7 @@
 
     MyExampleSoftwareChannelInstance()
         : m_input_channels(new EditableChannelIDListProperty())
-        , m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN() })
+        , m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN() })
     {
         // make property m_input_channels visible in the GUI
         m_input_channels->setVisiblity("PUBLIC");
@@ -128,7 +128,7 @@
         // Verify that all configured input channels can be used to compute an output
         const auto current_channel_ids = m_input_channels->getValue().m_values;
         bool is_valid = true;
-        if (current_channel_ids.size() != 2)
+        if (current_channel_ids.size() != 3)
         {
             is_valid = false;
         }
@@ -241,5 +241,5 @@
             return std::make_pair(context.m_channel_iterators[channel_id], getInputChannelProxy(channel_id)->getTimeBase().m_frequency);
         };
 
-        std::array<std::pair<odk::framework::StreamIterator, double>, 2> iterators = {
+        std::array<std::pair<odk::framework::StreamIterator, double>, 3> iterators = {
             prepare_iterator(0),
@@ -245,5 +245,6 @@
             prepare_iterator(0),
-            prepare_iterator(1)
+            prepare_iterator(1),
+            prepare_iterator(2)
         };
 
         std::uint64_t start_sample = odk::convertTimeToTickAtOrAfter(context.m_window.first,  m_timebase_frequency);
@@ -260,7 +261,7 @@
                 const double output_time = odk::convertTickToTime(sample_index, m_timebase_frequency);
 
                 // Read all input channels up until output_time (channels with slower sample rate might reuse the old value when no newer sample is available)
-                for (std::size_t channel_index = 0; channel_index < 2; ++channel_index)
+                for (std::size_t channel_index = 0; channel_index < 3; ++channel_index)
                 {
                     auto& iterator = iterators[channel_index].first;
                     const double timebase_frequency = iterators[channel_index].second;
@@ -276,7 +277,7 @@
             }
 
             // Read remaining samples to prevent missing samples in the next call of process
-            for (std::size_t channel_index = 0; channel_index < 2; ++channel_index)
+            for (std::size_t channel_index = 0; channel_index < 3; ++channel_index)
             {
                 auto& iterator = iterators[channel_index].first;
                 while (iterator.data())
@@ -291,7 +292,7 @@
             // no resampling required, read every sample from both input channels and compute an output sample
             for (auto sample_index = start_sample; sample_index < end_sample; ++sample_index)
             {
-                for (std::size_t channel_index = 0; channel_index < 2; ++channel_index)
+                for (std::size_t channel_index = 0; channel_index < 3; ++channel_index)
                 {
                     auto& iterator = iterators[channel_index].first;
                     if (iterator.valid())
@@ -324,7 +325,7 @@
      */
     ODK_NODISCARD inline double computeOutput() const
     {
-        return m_current_values[0] + m_current_values[1];
+        return m_current_values[0] + m_current_values[1] + m_current_values[2];
     }
 
     /**
@@ -350,7 +351,7 @@
 
     // move to base?
     std::shared_ptr<EditableChannelIDListProperty> m_input_channels;
-    std::array<double, 2> m_current_values;
+    std::array<double, 3> m_current_values;
     bool m_resampling_enabled = false;
     double m_timebase_frequency = 0.0;
 };
```

ravivelugula, 2024-11-18:
Hi, 

I have modified the as mentioned above, however, the same issue (crashing of oxygen) is occurring. See the modified code attached below. Suggest if any further modifications needed.

// Copyright DEWETRON GmbH 2019-2021

#include "odkfw_properties.h"
#include "odkfw_software_channel_plugin.h"
#include "odkapi_utils.h"

#include <array>
#include <cmath>
#include <string.h>

#define DONT_ENFORCE_SYNC_AND_ASYNC

static const char* PLUGIN_MANIFEST =
R"XML(<?xml version="1.0"?>
<OxygenPlugin name="ODK_SUMTHREE_CHANNELS" version="1.0" uuid="D9C295C0-CBB9-4412-9B4A-0C5B1ACF3GH6">
  <Info name="Example Plugin: Sumthree channels">
    <Vendor name="DEWETRON GmbH"/>
    <Description>SDK Example plugin that sums up the values of three input channels and writes it to the output channel</Description>
  </Info>
  <Host minimum_version="3.7"/>
</OxygenPlugin>
)XML";

static const char* TRANSLATION_EN =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="en" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_SUMTHREE_CHANNELS/MyInputChannels</source><translation>Input Channels</translation></message>
    </context>
</TS>
)XML";

static const char* TRANSLATION_DE =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="de" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_SUMTHREE_CHANNELS/MyInputChannels</source><translation>Eingangskan<byte value="xe4"/>le</translation></message>
    </context>
</TS>
)XML";

static const char* KEY_INPUT_CHANNELS = "ODK_SUMTHREE_CHANNELS/MyInputChannels";

using namespace odk::framework;

class MyExampleSoftwareChannelInstance : public SoftwareChannelInstance
{
public:

    MyExampleSoftwareChannelInstance()
        : m_input_channels(new EditableChannelIDListProperty())
        , m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN() })
    {
        // make property m_input_channels visible in the GUI
        m_input_channels->setVisiblity("PUBLIC");
    }

    /**
     * Return the information that is used to display the software channel in the calculation list in the GUI
     */
    static odk::RegisterSoftwareChannel getSoftwareChannelInfo()
    {
        odk::RegisterSoftwareChannel telegram;
        telegram.m_display_name = "Example Plugin: Sumthree channels";
        telegram.m_service_name = "AddSyncAsync";
        telegram.m_display_group = "Own Math";
[... 278 lines cut ...]
                return;
            }
        }

        m_resampling_enabled = false;
    }

    // move to base?
    std::shared_ptr<EditableChannelIDListProperty> m_input_channels;
    std::array<double, 3> m_current_values;
    bool m_resampling_enabled = false;
    double m_timebase_frequency = 0.0;
};

class MyExamplePlugin : public SoftwareChannelPlugin<MyExampleSoftwareChannelInstance>
{
public:
    void registerTranslations() final
    {
        addTranslation(TRANSLATION_EN);
        addTranslation(TRANSLATION_DE);
    }
};

OXY_REGISTER_PLUGIN1("ODK_SUMTHREE_CHANNELS", PLUGIN_MANIFEST, MyExamplePlugin);

ravivelugula, 2024-11-18:
Hi, now the sum of three channel plugin in working fine. 

Thank you

closed 2024-11-18 by ravivelugula