DEWETRON/OXYGEN-SDK #40: Issue while using cosine of a channel data

ravivelugula, 2024-11-18:
Hi, 

The below problem (see attached) is occurring while using cos [input channel] (see attached formula).

Kindly help me in resolving the issue

Thank you
[image]
[image]

ravivelugula, 2024-11-19:
Hi,

I want to calculate Vq i.e., V1 * cos(theta). The V1 and theta are the input channels. I have run the oxygen from visual studio debugger and the plugin developed successfully since it is noted that if select any other input channel (example I1) instead of theta, the plugin working properly and giving the result. The problem occurring only if I select theta input channel.

Kindly suggest suitable solution.

Thank you
[image]
[image]
[image]

glaure, 2024-11-19:
I tried the same setup with a modified sum channels plugin without any problem. 
Please provide the source code to your plugin.

Bye Gunther