DEWETRON/OXYGEN-SDK #41: Ld of PMSM calculation

ravivelugula, 2024-11-20:
Hi,

I am developing a plugin that calculated the Ld of PMSM using 5 input channels (see the attached formula below). I have modified the example sum of two channels. However, the plugin developed, the oxygen is crashing earlier mentioned. Will the example code can be used for 5 channels inputs upon modifying the code.

Kindly help me in this regard.
[image]

Thanks 
Ravi

glaure, 2024-11-20:
Please send me the complete source code of your plugin.

Best regards Gunther

ravivelugula, 2024-11-20:
Hi,

Please find the zip of source code. 

Thank you

ravivelugula, 2024-11-20:
// Copyright DEWETRON GmbH 2019-2021

#include "odkfw_properties.h"
#include "odkfw_software_channel_plugin.h"
#include "odkapi_utils.h"

#include <array>
#include <cmath>
#include <string.h>

#define DONT_ENFORCE_SYNC_AND_ASYNC

static const char* PLUGIN_MANIFEST =
R"XML(<?xml version="1.0"?>
<OxygenPlugin name="ODK_LD_CHANNELS" version="1.0" uuid="D9C295C0-CBB9-4412-9B4A-0C5B1ACF3EP6">
  <Info name="Example Plugin: Ld channels">
    <Vendor name="DEWETRON GmbH"/>
    <Description>SDK Example plugin that Ld up the values of two input channels and writes it to the output channel</Description>
  </Info>
  <Host minimum_version="3.7"/>
</OxygenPlugin>
)XML";

static const char* TRANSLATION_EN =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="en" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_LD_CHANNELS/MyInputChannels</source><translation>Input Channels</translation></message>
    </context>
</TS>
)XML";

static const char* TRANSLATION_DE =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="de" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_LD_CHANNELS/MyInputChannels</source><translation>Eingangskan<byte value="xe4"/>le</translation></message>
    </context>
</TS>
)XML";

static const char* KEY_INPUT_CHANNELS = "ODK_LD_CHANNELS/MyInputChannels";
static const char* KEY_RESI_ANCE = "Resistance";

using namespace odk::framework;

class MyExampleSoftwareChannelInstance : public SoftwareChannelInstance
{
public:

    MyExampleSoftwareChannelInstance()
        : m_input_channels(new EditableChannelIDListProperty())
        , m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN() })
        , m_resi_ance(std::make_shared<EditableUnsignedProperty>(0.9, 0.1, 10))
    {
        // make property m_input_channels visible in the GUI
        m_input_channels->setVisiblity("PUBLIC");
        m_resi_ance->setVisiblity("PUBLIC");
    }

    /**
     * Return the information that is used to display the software channel in the calculation list in the GUI
     */
    static odk::RegisterSoftwareChannel getSoftwareChannelInfo()
    {
        odk::RegisterSoftwareChannel telegram;
        telegram.m_display_name = "Example Plugin: Ld channels";
        telegram.m_service_name = "AddSyncAsync";
        telegram.m_display_group = "Own Math";
        telegram.m_description = "Adds a channel that calculates the ld using five input channels.";
[... 282 lines cut ...]
            }
        }

        m_resampling_enabled = false;
    }

    // move to base?
    std::shared_ptr<EditableChannelIDListProperty> m_input_channels;
    std::array<double, 5> m_current_values;
    std::shared_ptr<EditableUnsignedProperty> m_resi_ance;
    bool m_resampling_enabled = false;
    double m_timebase_frequency = 0.0;
};

class MyExamplePlugin : public SoftwareChannelPlugin<MyExampleSoftwareChannelInstance>
{
public:
    void registerTranslations() final
    {
        addTranslation(TRANSLATION_EN);
        addTranslation(TRANSLATION_DE);
    }
};

OXY_REGISTER_PLUGIN1("ODK_LD_CHANNELS", PLUGIN_MANIFEST, MyExamplePlugin);``

glaure, 2024-11-20:
> This email originated outside of DEWETRON. DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

Hi,

some of the source code got removed by the html filter.

When I fill in the missing stuff, the plugin works without any issue.

Please try to post your source code to the github issue. There is a separate option
For source code in the editor.

Bye Gunther

<https://www.dewetron.com>[cid:dewetron-logo_email-signature-150_ce539afa-b6be-4be4-9497-7b265fb4b6e5.png]<https://www.dewetron.com/> <https://www.my-company.com/>
Gunther Laure | Team Leader Software Development
***@***.*** | www.dewetron.com<https://www.dewetron.com>
+433163070284
DEWETRON, Parkring 4, 8074 Grambach, AUSTRIA
<https://www.facebook.com/MyCompany>[cid:linkedin_a56b72d3-e383-4d44-be3b-70b0c715225d.png]<https://www.linkedin.com/company/dewetron/>  [cid:youtube_880b6f0d-8a12-4051-a2ea-c74b7beafd50.png] <https://www.youtube.com/channel/UCE76NeVVgWF65etH8E1bOog>   [cid:facebook_79c45449-903c-4ec0-b195-ff7161baa5ad.png] <https://www.facebook.com/dewetron.gmbh>   [cid:instagram_3fe69145-c3ef-4d06-9c69-a7d5c15df8c5.png] <https://www.instagram.com/wearedewetron/>   [cid:github_281fdf2e-da25-44d8-99d8-3c2dea93a9ea.png] <https://github.com/DEWETRON>

<http://bit.ly/dewetron-news>

<https://www.dewetron.com/products/litepa-standard-power-analyzer/>[cid:litepa-power-analyzer_en_b7044e47-eae1-46f0-8a80-af6ec65c5496.png]<https://www.dewetron.com/products/litepa-standard-power-analyzer/>

#include

ConfigKeys
ODK_LD_CHANNELS/MyInputChannelsInput Channels

ConfigKeys
ODK_LD_CHANNELS/MyInputChannelsEingangskanle

ravivelugula, 2024-11-20:
`// Copyright DEWETRON GmbH 2019-2021

#include "odkfw_properties.h"
#include "odkfw_software_channel_plugin.h"
#include "odkapi_utils.h"

#include <array>
#include <cmath>
#include <string.h>

#define DONT_ENFORCE_SYNC_AND_ASYNC

static const char* PLUGIN_MANIFEST =
R"XML(<?xml version="1.0"?>
<OxygenPlugin name="ODK_LD_CHANNELS" version="1.0" uuid="D9C295C0-CBB9-4412-9B4A-0C5B1ACF3EP6">
  <Info name="Example Plugin: Ld channels">
    <Vendor name="DEWETRON GmbH"/>
    <Description>SDK Example plugin that Ld up the values of two input channels and writes it to the output channel</Description>
  </Info>
  <Host minimum_version="3.7"/>
</OxygenPlugin>
)XML";

static const char* TRANSLATION_EN =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="en" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_LD_CHANNELS/MyInputChannels</source><translation>Input Channels</translation></message>
    </context>
</TS>
)XML";

static const char* TRANSLATION_DE =
R"XML(<?xml version="1.0"?>
<TS version="2.1" language="de" sourcelanguage="en">
    <context><name>ConfigKeys</name>
        <message><source>ODK_LD_CHANNELS/MyInputChannels</source><translation>Eingangskan<byte value="xe4"/>le</translation></message>
    </context>
</TS>
)XML";

static const char* KEY_INPUT_CHANNELS = "ODK_LD_CHANNELS/MyInputChannels";
static const char* KEY_RESI_ANCE = "Resistance";

using namespace odk::framework;

class MyExampleSoftwareChannelInstance : public SoftwareChannelInstance
{
public:

    MyExampleSoftwareChannelInstance()
        : m_input_channels(new EditableChannelIDListProperty())
        , m_current_values({ std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN(), std::numeric_limits<double>::quiet_NaN() })
        , m_resi_ance(std::make_shared<EditableUnsignedProperty>(0.9, 0.1, 10))
    {
        // make property m_input_channels visible in the GUI
        m_input_channels->setVisiblity("PUBLIC");
        m_resi_ance->setVisiblity("PUBLIC");
    }

    /**
     * Return the information that is used to display the software channel in the calculation list in the GUI
     */
    static odk::RegisterSoftwareChannel getSoftwareChannelInfo()
    {
        odk::RegisterSoftwareChannel telegram;
        telegram.m_display_name = "Example Plugin: Ld channels";
        telegram.m_service_name = "AddSyncAsync";
        telegram.m_display_group = "Own Math";
        telegram.m_description = "Adds a channel that calculates the ld using five input channels.";
[... 282 lines cut ...]
            }
        }

        m_resampling_enabled = false;
    }

    // move to base?
    std::shared_ptr<EditableChannelIDListProperty> m_input_channels;
    std::array<double, 5> m_current_values;
    std::shared_ptr<EditableUnsignedProperty> m_resi_ance;
    bool m_resampling_enabled = false;
    double m_timebase_frequency = 0.0;
};

class MyExamplePlugin : public SoftwareChannelPlugin<MyExampleSoftwareChannelInstance>
{
public:
    void registerTranslations() final
    {
        addTranslation(TRANSLATION_EN);
        addTranslation(TRANSLATION_DE);
    }
};

OXY_REGISTER_PLUGIN1("ODK_LD_CHANNELS", PLUGIN_MANIFEST, MyExamplePlugin);`

ravivelugula, 2024-11-20:
Hi,

I have added the source code in "code" GitHub. Let me know if further information is needed to provide.

Thank you

closed 2024-11-21 by ravivelugula